esp-rs/esp-template #2: wish: Blinking LED

stappersg, 2022-04-24:
The current template is a _hollow shell_. My wish is that it gets _"blinking LED"_ code.

Benefits I see:
* The whole toolchain can be verified
* There is something that works and can be tweaked ( frequency of LED blinking )

Possible down side:  The unwanted source code has to be removed.
(but that is outweighted by all project starts that do need / want something that proves a working target)

jessebraham, 2022-04-26:
I guess we need to make a decision regarding the intended purpose of this template. There are already examples in `esp-hal`, blinky included, so I had created this template to generate the bare minimum required to get a properly configured project. I'm not entirely against modifying this template to reflect that example, but I will need to think about it.

MabezDev, 2022-04-26:
I would say that HAL examples (and other external examples) should be used to verify the toolchain. This template is deliberately hollow, as it's meant as a starting point for a project.