esp-rs/esp-template #21: Help - erro on generate template

samuel-cavalcanti, 2022-08-11:
Hi, I'm trying to use cargo generate, but it throw this error:
```zsh
cargo generate https://github.com/esp-rs/esp-template          
error: Found argument 'https://github.com/esp-rs/esp-template' which wasn't expected, or isn't valid in this context

USAGE:
    cargo generate [FLAGS] [OPTIONS] --git <git>

For more information try --help
```

when I use  --git:
```zsh
cargo generate --git=https://github.com/esp-rs/esp-template
 Project Name: test-esp8266
 Creating project called `test-esp8266`...
Error:  Git Error: failed to clone into: /home/samuel/Downloads/test-esp8266IXEFXF
```
Have I did something wrong ?

jessebraham, 2022-08-12:
Please make sure your version of `cargo-generate` is up to date, I'm running `0.16.0`. Not sure why you're seeing this error, I just ran the same command and it worked fine:

```bash
jesse@mbp ~
λ cargo generate --vcs none https://github.com/esp-rs/esp-template
⚠️   Favorite `https://github.com/esp-rs/esp-template` not found in config, using it as a git repository: https://github.com/esp-rs/esp-template
🤷   Project Name : foo
🔧   Destination: /Users/jesse/foo ...
🔧   Generating template ...
✔ 🤷   Configure project to use Dev Containers (VS Code, GitHub Codespaces and Gitpod)? · false
✔ 🤷   Which MCU to target? · esp32
[ 1/11]   Done: .cargo/config.toml
[ 2/11]   Done: .cargo
[ 3/11]   Done: .gitignore
[ 4/11]   Done: .vscode/settings.json
[ 5/11]   Done: .vscode
[ 6/11]   Done: Cargo.toml
[ 7/11]   Done: LICENSE-APACHE
[ 8/11]   Done: LICENSE-MIT
[ 9/11]   Done: rust-toolchain.toml
[10/11]   Done: src/main.rs
[11/11]   Done: src
🔧   Moving generated files into: `/Users/jesse/foo`...
✨   Done! New project created /Users/jesse/foo
```

We do not currently support the ESP8266 mostly just because it's not a part of the `esp-hal` mono-repository, and probably never will be. It does not share common peripheral hardware like the other ESP32* chips do. It should be possible to add an option for the ESP8266 which pulls in `esp8266-hal`, though.

samuel-cavalcanti, 2022-08-12:
You are right, I have used  an old version of cargo generate. 

I have an esp8266 and I'm trying to use esp-rs, Do you know where can I get some tips or started point ?

jessebraham, 2022-08-12:
Unfortunately [esp8266-hal](https://github.com/esp-rs/esp8266-hal) isn't really actively maintained anymore, but there are some basic examples there. There is no support for Wi-Fi but most of the other peripherals have some level of support. You won't be able to use `esp-template` but the required configuration files are all present in the repository, so you should be able to just copy them.

samuel-cavalcanti, 2022-08-12:
thank you  @jessebraham  👍🏽

closed 2022-08-12 by samuel-cavalcanti