esp-rs/esp-template #43: esp32c3 can't boot

Dreagonmon, 2023-01-06:
Board: https://cdn.openluat-luatcommunity.openluat.com/attachment/CORE-ESP32-C3%E8%AE%BE%E8%AE%A1%E5%8E%9F%E7%90%86%E5%9B%BE_V1.2.pdf

Toolchains installed by espup

project generated using `cargo generate --git https://github.com/esp-rs/esp-template`, target esp32c3.

Output:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xe (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x172c
load:0x3ce00000,len:0x92840
load:0x50285e28,len:0xfeaa50a8
Invalid image block, can't boot.
ets_main.c 333 
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xe (SPI_FAST_FLASH_BOOT)
Saved PC:0x40047ed2
0x40047ed2 - _stack_start
    at ??:??
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x172c
load:0x3ce00000,len:0x92840
load:0x50285e28,len:0xfeaa50a8
Invalid image block, can't boot.
ets_main.c 333
```

This board using a micropython firmware with custom partition table previously:
```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 0x2F0000,
vfs,      data, fat,     0x300000, 0x100000,
```

I tried to generate another project with esp32 target and esp32 board, it works fine.

wirano, 2023-05-19:
I use espflash v2.0.0-rc.3 solved this problem, this may help for you :)

SergioGasquez, 2023-06-23:
`espflash v2.0.0` is now officially released, I'll be closing the issue

closed 2023-06-23 by SergioGasquez