esp-rs/esp-template #60: bug: cargo generate fails: Runtime error: System command `cargo fmt` returned non-zero status: exit status: 1

ilka-schulz, 2023-03-13:
I tried to generate a new project according to the README but it fails:

```text
$ cargo generate -a esp-rs/esp-template
⚠️   Favorite `esp-rs/esp-template` not found in config, using it as a git repository: https://github.com/esp-rs/esp-template.git
🤷   Project Name: no_std_test
⚠️   Renaming project called `no_std_test` to `no-std-test`...
🔧   Destination: /home/user/no-std-test ...
🔧   project-name: no-std-test ...
🔧   Generating template ...
✔ 🤷   Which MCU to target? · esp32
✔ 🤷   Enable allocations via the esp-alloc crate? · true
✔ 🤷   Configure project to use Dev Containers (VS Code, GitHub Codespaces and Gitpod)? · true
[ 1/27]   Done: .cargo/config.toml                                                                                                                                
[ 2/27]   Done: .cargo                                                                                                                                            
[ 3/27]   Done: .devcontainer/Dockerfile                                                                                                                          
[ 4/27]   Done: .devcontainer/devcontainer.json                                                                                                                   
[ 5/27]   Done: .devcontainer                                                                                                                                     
[ 6/27]   Done: .dockerignore                                                                                                                                     
[ 7/27]   Done: .gitignore                                                                                                                                        
[ 8/27]   Done: .gitpod.Dockerfile                                                                                                                                
[ 9/27]   Done: .gitpod.yml                                                                                                                                       
[10/27]   Done: .vscode/launch.json                                                                                                                               
[11/27]   Done: .vscode/settings.json                                                                                                                             
[12/27]   Done: .vscode/tasks.json                                                                                                                                
[13/27]   Done: .vscode                                                                                                                                           
[14/27]   Done: Cargo.toml                                                                                                                                        
[15/27]   Done: LICENSE-APACHE                                                                                                                                    
[16/27]   Done: LICENSE-MIT                                                                                                                                       
[17/27]   Done: docs/README.md                                                                                                                                    
[18/27]   Done: docs                                                                                                                                              
[19/27]   Ignored: post-script.rhai                                                                                                                               
[20/27]   Ignored: pre-script.rhai                                                                                                                                
[21/27]   Done: rust-toolchain.toml                                                                                                                               
[22/27]   Done: scripts/build.sh                                                                                                                                  
[23/27]   Done: scripts/flash.sh                                                                                                                                  
[24/27]   Done: scripts/run-wokwi.sh                                                                                                                              
[25/27]   Done: scripts                                                                                                                                           
[26/27]   Done: src/main.rs                                                                                                                                       
[27/27]   Done: src                                                                                                                                               
Error: ⛔   Failed executing script: post-script.rhai

Caused by:
    Runtime error: System command `cargo fmt` returned non-zero status: exit status: 1
```

jessebraham, 2023-03-13:
Works fine on my system using the same configuration as the output you've posted. Make sure that `cargo-generate` is up to date I suppose, not really sure what else to recommend.

pixelskull, 2023-03-18:
Had the same problem, it got fixed when uninstalling rust and reinstalling following [the book](https://esp-rs.github.io/book/installation/installation.html). Still no idea what went wrong. Probably something

SergioGasquez, 2023-04-03:
Do you have `rustfmt` component installed? If not, try doing `rustup component add rustfmt` before running the cargo-generate command

SergioGasquez, 2023-06-23:
I'll close the issue since its not active and we no longer require `rustfmt`, see https://github.com/esp-rs/esp-template/pull/91

closed 2023-06-23 by SergioGasquez