esp-rs/esp-template #61: espflash::connection_failed when building

drdozer, 2023-03-15:
I have an esp32-cam-mb (an esp32-cam with a programming mb) and I am able to build and run the idf template (https://github.com/esp-rs/esp-idf-template) out of the box without any alteration. When building the esp-template, I get an error:

```
~/devel/turingatemyhamster/scratch/cam32-baremetal-blink$ cargo run
    Finished dev [unoptimized + debuginfo] target(s) in 0.07s
     Running `espflash flash --monitor target/xtensa-esp32-none-elf/debug/cam32_baremetal_blink`
New version of espflash is available: v2.0.0-rc.3

Error: espflash::connection_failed

  × Error while connecting to device
  ╰─▶ Serial port not found
  help: Ensure that the device is connected and your host recognizes the serial adapter
```

When I edited the file `.cargo/config.toml` to replace `runner = "espflash flash --monitor"` with `runner = "espflash --monitor"`, it succeeded in uploading and running code, although it appears there's some other issue that prevents it reaching the "hello world!" line, but I guess that's another bug report.

For completeness, here is the successful session with esp-idf-template.
```
~/devel/turingatemyhamster/scratch/cam32-idf-blink$ cargo run
    Finished dev [optimized + debuginfo] target(s) in 0.12s
     Running `espflash --monitor target/xtensa-esp32-espidf/debug/cam32-idf-blink`
New version of espflash is available: v2.0.0-rc.3

Serial port: /dev/ttyUSB0
Connecting...

Chip type:         ESP32 (revision 3)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi, BT, Dual Core, 240MHz, Coding Scheme None
MAC address:       c8:f0:9e:4d:4c:34
App/part. size:    449296/4128768 bytes, 10.88%
[00:00:01] ########################################      16/16      segment 0x1000                                  [00:00:00] ########################################       1/1       segment 0x8000                                  [00:00:23] ########################################     228/228     segment 0x10000                                 
Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ets Jul 29 2019 12:21:46
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0048,len:12
ho 0 tail 12 room 4
load:0x3fff0054,len:4800
load:0x40078000,len:17448
0x40078000 - __udivmoddi4
    at ??:??
load:0x4007c428,len:4840
0x4007c428 - __udivmoddi4
    at ??:??
entry 0x4007c6a0
0x4007c6a0 - __udivmoddi4
    at ??:??
I (227) cpu_start: Pro cpu up.
I (227) cpu_start: Starting app cpu, entry point is 0x40081164
0x40081164 - call_start_cpu1
    at /home/nmrp3/devel/turingatemyhamster/scratch/cam32-idf-blink/.embuild/espressif/esp-idf/release-v4.4/components/esp_system/port/cpu_start.c:148
I (0) cpu_start: App cpu up.
I (241) cpu_start: Pro cpu start user code
I (241) cpu_start: cpu freq: 160000000
I (241) cpu_start: Application information:
I (246) cpu_start: Project name:     libespidf
I (251) cpu_start: App version:      1
I (255) cpu_start: Compile time:     Mar 13 2023 14:24:44
I (261) cpu_start: ELF file SHA256:  0000000000000000...
I (267) cpu_start: ESP-IDF:          fee50c2-dirty
I (273) cpu_start: Min chip rev:     v0.0
I (277) cpu_start: Max chip rev:     v3.99 
I (282) cpu_start: Chip rev:         v3.0
I (287) heap_init: Initializing. RAM available for dynamic allocation:
I (294) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (300) heap_init: At 3FFB23A8 len 0002DC58 (183 KiB): DRAM
I (306) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (313) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (319) heap_init: At 4008AB20 len 000154E0 (85 KiB): IRAM
I (327) spi_flash: detected chip: generic
I (330) spi_flash: flash io: dio
I (335) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Hello, world!
```

miathedev, 2023-04-01:
Surely not the answer your looking for.

But it looks like it worked or doesnt it? . The Hello World line is printed into your log as well.
Or what do you mean with: "_although it appears there's some other issue that prevents it reaching the "hello world!"_?

____
I can confirm, your workaround fixed it for me too.
I added a blinking led example as addition:
```
#![no_std]
#![no_main]

extern crate alloc;
use esp_backtrace as _;
use esp_println::println;
use hal::{clock::ClockControl, peripherals::Peripherals, prelude::*, timer::TimerGroup, Rtc, Delay, IO};
#[global_allocator]
static ALLOCATOR: esp_alloc::EspHeap = esp_alloc::EspHeap::empty();

fn init_heap() {
    const HEAP_SIZE: usize = 32 * 1024;

    extern "C" {
        static mut _heap_start: u32;
    }

    unsafe {
        let heap_start = &_heap_start as *const _ as usize;
        ALLOCATOR.init(heap_start as *mut u8, HEAP_SIZE);
    }
}

#[entry]
fn main() -> ! {
    init_heap();
    let peripherals = Peripherals::take();
    let system = peripherals.SYSTEM.split();
    let clocks = ClockControl::boot_defaults(system.clock_control).freeze();

    // Disable the RTC and TIMG watchdog timers
    let mut rtc = Rtc::new(peripherals.RTC_CNTL);
    let timer_group0 = TimerGroup::new(peripherals.TIMG0, &clocks);
    let mut wdt0 = timer_group0.wdt;
    let timer_group1 = TimerGroup::new(peripherals.TIMG1, &clocks);
    let mut wdt1 = timer_group1.wdt;
    rtc.swd.disable();
    rtc.rwdt.disable();
    wdt0.disable();
    wdt1.disable();
    println!("Hello world!");

    let io = IO::new(peripherals.GPIO, peripherals.IO_MUX);
    let mut led = io.pins.gpio5.into_push_pull_output();

    led.set_high().unwrap();

    // Initialize the Delay peripheral, and use it to toggle the LED state in a
    // loop.
    let mut delay = Delay::new(&clocks);

    loop {
        led.toggle().unwrap();
        delay.delay_ms(500u32);
    }
}
```

bjoernQ, 2023-04-03:
Just to explain things a bit:

espflash 2.x needs the `flash` command while espflash 1.x will just try to use the command as the name of the serial port.

You can check your version of espflash via
```
❯ espflash --version
espflash 2.0.0-rc.3
```

The situation will be much better once espflash 2.0 is the official current version.

drdozer, 2023-04-03:
> espflash 2.x needs the `flash` command while espflash 1.x will just try to use the command as the name of the serial port.

Thanks. I had not realised this initially, and was also thrown by the two templates being different in the espflash commands they used. I did submit a patch to add commands for both espflash 1 and 2, with one version commented out to help noobs understand what's going on while we wait for espflash2 to become standard. I am happy for this issue to be closed since it is resolved for me.

SergioGasquez, 2023-06-22:
`espflash` 2.0 is now released, closing the issue!

closed 2023-06-22 by SergioGasquez