esp-rs/esp-template #62: fails to boot

drdozer, 2023-03-15:
When I `cargo run` this template, it successfully uploads the image but the chip fails to boot up.

```
~/devel/turingatemyhamster/scratch/cam32-baremetal-blink$ cargo run
    Finished dev [unoptimized + debuginfo] target(s) in 0.08s
     Running `espflash --monitor target/xtensa-esp32-none-elf/debug/cam32_baremetal_blink`
New version of espflash is available: v2.0.0-rc.3

Serial port: /dev/ttyUSB0
Connecting...

Chip type:         ESP32 (revision 3)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi, BT, Dual Core, 240MHz, Coding Scheme None
MAC address:       c8:f0:9e:4d:4c:34
App/part. size:    150224/4128768 bytes, 3.64%
[00:00:01] ########################################      16/16      segment 0x1000                                  [00:00:00] ########################################       1/1       segment 0x8000                                  [00:00:05] ########################################      55/55      segment 0x10000                                 
Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ets Jul 29 2019 12:21:46
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0048,len:12
ho 0 tail 12 room 4
load:0x3fff0054,len:4800
load:0x40078000,len:17448
0x40078000 - ets_delay_us
    at ??:??
load:0x4007c428,len:4840
0x4007c428 - ets_delay_us
    at ??:??
entry 0x4007c6a0
0x4007c6a0 - ets_delay_us
    at ??:??
ets Jul 29 2019 12:21:46
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0048,len:12
ho 0 tail 12 room 4
load:0x3fff0054,len:4800
load:0x40078000,len:17448
0x40078000 - ets_delay_us
    at ??:??
load:0x4007c428,len:4840
0x4007c428 - ets_delay_us
    at ??:??
entry 0x4007c6a0
0x4007c6a0 - ets_delay_us
    at ??:??
<loops for ever>
```

The esp-idf-template project runs out of the box. The output is below for comparison. 

```
~/devel/turingatemyhamster/scratch/cam32-idf-blink$ cargo run
    Finished dev [optimized + debuginfo] target(s) in 0.12s
     Running `espflash --monitor target/xtensa-esp32-espidf/debug/cam32-idf-blink`
New version of espflash is available: v2.0.0-rc.3

Serial port: /dev/ttyUSB0
Connecting...

Chip type:         ESP32 (revision 3)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi, BT, Dual Core, 240MHz, Coding Scheme None
MAC address:       c8:f0:9e:4d:4c:34
App/part. size:    449296/4128768 bytes, 10.88%
[00:00:01] ########################################      16/16      segment 0x1000                                  [00:00:00] ########################################       1/1       segment 0x8000                                  [00:00:23] ########################################     228/228     segment 0x10000                                 
Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ets Jul 29 2019 12:21:46
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0048,len:12
ho 0 tail 12 room 4
load:0x3fff0054,len:4800
load:0x40078000,len:17448
0x40078000 - __udivmoddi4                                                                                           
    at ??:??                                                                                                        
load:0x4007c428,len:4840
0x4007c428 - __udivmoddi4                                                                                           
    at ??:??                                                                                                        
entry 0x4007c6a0
0x4007c6a0 - __udivmoddi4                                                                                           
    at ??:??                                                                                                        
I (227) cpu_start: Pro cpu up.
I (227) cpu_start: Starting app cpu, entry point is 0x40081164
0x40081164 - call_start_cpu1                                                                                        
    at /home/nmrp3/devel/turingatemyhamster/scratch/cam32-idf-blink/.embuild/espressif/esp-idf/release-v4.4/components/esp_system/port/cpu_start.c:148                                                                                  
I (0) cpu_start: App cpu up.
I (241) cpu_start: Pro cpu start user code
I (241) cpu_start: cpu freq: 160000000
I (241) cpu_start: Application information:
I (246) cpu_start: Project name:     libespidf
I (251) cpu_start: App version:      1
I (255) cpu_start: Compile time:     Mar 13 2023 14:24:44
I (261) cpu_start: ELF file SHA256:  0000000000000000...
I (267) cpu_start: ESP-IDF:          fee50c2-dirty
I (273) cpu_start: Min chip rev:     v0.0
I (277) cpu_start: Max chip rev:     v3.99 
I (282) cpu_start: Chip rev:         v3.0
I (287) heap_init: Initializing. RAM available for dynamic allocation:
I (294) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (300) heap_init: At 3FFB23A8 len 0002DC58 (183 KiB): DRAM
I (306) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (313) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (319) heap_init: At 4008AB20 len 000154E0 (85 KiB): IRAM
I (327) spi_flash: detected chip: generic
I (330) spi_flash: flash io: dio
I (335) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Hello, world!
```

pixelskull, 2023-03-18:
Did you manage to get the template to run?

drdozer, 2023-03-19:
Not so far. I tried changing fairly random things but no dice. I tried copying over chunks of the config rom the idf template, but that obviously broke when trying to link.

MabezDev, 2023-03-19:
This is https://github.com/esp-rs/rust/issues/158, which has been fixed with https://github.com/esp-rs/esp-hal/pull/423 but has not yet been released on crates.io.

drdozer, 2023-03-19:
> This is esp-rs/rust#158 <https://github.com/esp-rs/rust/issues/158>,
> which has been fixed with esp-rs/esp-hal#423
> <https://github.com/esp-rs/esp-hal/pull/423> but has not yet been
> released on crates.io.

Cheers. I will test again once the update is released.

drdozer, 2023-03-23:
As an FYI, by adding the following to Cargo.toml, the project runs as expected.

```
[profile.release]
opt-level = "s"

[profile.dev]
debug = true # Symbols are nice and they don't increase the size on Flash
opt-level = "z"
```

So this is consistent with it being the bug for `opt-level=0`.

naschidaniel, 2023-03-26:
Thank you @drdozer. Your Information worked for me.

SergioGasquez, 2023-03-29:
Closed, since this should be fixed in 1.68.0

closed 2023-03-29 by SergioGasquez