esp-rs/esp-template #63: Cannot run `cargo test` using fresh template

teotwaki, 2023-03-20:
Hi,

This may be self-evident, and if it is I apologise. It appears that one cannot use `cargo test` using a fresh `cargo generate`d project based on this template:

```
$ cargo generate --git https://github.com/esp-rs/esp-template
[...]
🔧   Generating template ...
✔ 🤷   Enable allocations via the esp-alloc crate? · true
✔ 🤷   Which MCU to target? · esp32
✔ 🤷   Configure project to use Dev Containers (VS Code, GitHub Codespaces and Gitpod)? · false
[...]
$ cd foo
$ cargo test
[...]
   Compiling foo v0.1.0 (/Users/slau/src/unticks/foo)
error[E0463]: can't find crate for `test`

For more information about this error, try `rustc --explain E0463`.
error: could not compile `foo` due to previous error
```

Not sure if this is expected, intended, or unfixable.

MabezDev, 2023-03-20:
Hi, thanks for the issue! Testing on `no_std` is pretty target specific, so there isn't a general solution. The most common pattern I see is running tests on the host machine instead. This of course requires writing tests in a certain way (i.e, can't rely on hardware features of the chip). I know this doesn't help immediately but hopefully gives you some pointers to get a test setup working for you :). If you do find a good pattern, that would work in the template, would you mind submitting a PR?

teotwaki, 2023-03-20:
Hi!

I understand that testing on device is very specific, however I guess my question was more: why is the `#[test]` macro broken in projects generated using the template? I have a [`no_std` library](https://github.com/Unticks/arducam-mega/blob/main/src/lib.rs) where tests work fine.

Is there a specific thing that is done in the template which prevents the use of `#[test]`? Is this by design or a known side-effect?

jessebraham, 2023-03-20:
I would guess it's because there's no `.cargo/config.toml` file in that library, but that's just a guess.

MabezDev, 2023-03-21:
Library crates can be compiled for any arch (provided they don't contain any target-specific asm), this isn't generally the case for binary crates, which this template produces. In general, I see folks splitting their code into lib and bin, and adding testing in the lib crate, like you did for yours above.

elpiel, 2023-05-02:
I would like to join in this discussion as well.
Lately I've been running tests under linux and there is a way to support cargo test.

On the flip side, I'm getting an issue with esp-hal-common and needs upstream fixing to make it work.
Would be nice to set up defmt_test too for the template to run tests on the device

closed 2024-11-14 by jessebraham