esp-rs/esp-template #89: Meaning behind esp-template question "Use template default values" is not obvious.

georgik, 2023-05-31:
Current questions: Use template default values? true/false

Thr question not provide insight into what is the user selecting.

Recommended change: Use minimalist project profile? yes/no
In case of no, continue with configuration options.

Why is the "default" question not good? Because "default" can mean something else to different people. Default in my case would mean full-blown project with all the bells and whistles, so that I can easily integrate it with all the things used by app developers including CI with containers, Simulation and Debugging.

SergioGasquez, 2023-06-23:
Hi! Do you think updating the prompt to: `Use template default values? Using default values will set a minimal project configuration.` will be better?

georgik, 2023-06-23:
What about: `Use minimal project configuration?` Options: yes/no and yes might be set as default answer

MabezDev, 2023-06-24:
Throwing another option into the hat, what about flipping the question to:

Advanced options? yes/no, the default being no.

SergioGasquez, 2023-06-26:
> Throwing another option into the hat, what about flipping the question to:
> 
> Advanced options? yes/no, the default being no.

I like this approach, it would require changing all the CI that are building the template since it will require changing the placeholder name and default value, but it might be still worth

closed 2023-06-27 by SergioGasquez