esp-rs/esp-template #90: Lack of auto completion, proc macro expansion, etc

leungyauming, 2023-06-16:
I had my hands on a ESP32-Pico-Kit lately. I wanted to develop `no_std` apps on it. I am using Visual Studio Code with the rust-analyzer extension.

Firstly, auto completion is basically useless. For example, `peripherals.DPORT.split()`, I would get anything after I typed `peripherals.DPORT.` and hovering also doesn't show anything. I was wondering if it has anything to do with proc macro expansion (see below).

Secondly, rust-analyzer also pops an error stating that the `#[entry]` macro cannot be expanded. At first, it is `rust-analyzer-proc-macro-srv` not found, so I compiled one myself. Then it show `unsupported ABI: ...`, I then "solved" it by compiling `proc-macro-srv-cli` with the `sysroot-abi` feature enabled (although I'm not so sure if this is really the solution). However, I notice that recent Rust toolchains ship with a `rust-analyzer-proc-macro-srv` to solve the ABI issue when the `rust-analyzer` client itself is not built under the same version of toolchain, maybe I should file an issue over `esp-rs/rust`?

Even after there is no errors or whatsoever, auto completion and most language features are basically unusable.

I want to know if this is a common experience or not. I tried my best to solve it myself but I am basically stuck now...

BastiDood, 2023-06-18:
Regarding the `#[entry]` proc-macro, esp-rs/rust-build#215 might be able to fix that.

SergioGasquez, 2023-06-19:
See https://github.com/esp-rs/espup/issues/254

closed 2023-06-19 by SergioGasquez