esp-rs/esp-template #95: Feature request: include CI files

Ben-PH, 2023-06-30:
I've started a project based on generating from this template, and setting up CI would be made easier if the generated files included some CI. This is what I've put together for my project:

```yml
name: check

on:
  pull_request:
    branches:
      - main

jobs:
  lint:
    name: Lint
    runs-on: ubuntu-22.04
    env:
      CARGO_TERM_COLOR: always

    steps:
      - uses: actions/checkout@v3
      - uses: esp-rs/xtensa-toolchain@v1.5
      - uses: Swatinem/rust-cache@v2

      - name: check rustfmt
        run: cargo fmt -- --check --color always

      - name: clippy
        run: cargo clippy -- -D warnings
```

the steps were partially based on the CI files for this repo.

If there's a way to bring in the right toolchain in a way that is nice and quick, and only sets things up for the appropriate platform (I'm using `S3`, but I notice work is being done to setup for non `S3` platforms), that would probably be ideal.

SergioGasquez, 2023-06-30:
I see some value in adding this as a prompt now that we have the `advanced` prompt, but that would mean that the project becomes more complex.  I would like to know other's opinion in adding this @esp-rs/espressif 

> If there's a way to bring in the right toolchain in a way that is nice and quick, and only sets things up for the appropriate platform (I'm using S3, but I notice work is being done to setup for non S3 platforms), that would probably be ideal.

The Xtensa Rust toolchain can be set with `xtensa-toolchain` action as you are doing, and use the [`buildtargets` argument](https://github.com/esp-rs/xtensa-toolchain#inputs) if you only want to install it for some targets.

Ben-PH, 2023-06-30:
Thanks for that link, I'll follow those docs :)

bjoernQ, 2023-06-30:
I think this would be a good addition. But I totally agree this should be added in a way that doesn't end up in maintenance hell

SergioGasquez, 2023-07-05:
Hi! Just did some testing to implement this feature, see changes in https://github.com/esp-rs/esp-template/compare/main...SergioGasquez:esp-template:feat/ci a few notes:
- It shouldn't be too difficult to maintain since it's behind the `advanced prompt`
- Sadly, [cargo-generate does not support wildcards in the `ignore`](https://cargo-generate.github.io/cargo-generate/templates/ignoring.html), hence I have to list all the files under `.github`
  - This is ugly but no big deal
- The problem comes with the `rust_ci.yml`, cargo-generate is not able to properly generate the file because GH Action uses the same syntax for variables (`{{ <variable_name> }}`), and if we declare any GH actions variable the file won't be parsed.
  - I've opened https://github.com/cargo-generate/cargo-generate/issues/976, maybe there is a workaround that I can't think of.

SergioGasquez, 2023-07-05:
> The problem comes with the `rust_ci.yml`, cargo-generate is not able to properly generate the file because GH Action uses the same syntax for variables (`{{ <variable_name> }}`), and if we declare any GH actions variable the file won't be parsed.

Fixed as suggested in the cargo-generate issues using [`raw` tag](https://shopify.dev/docs/api/liquid/tags/raw). Will open a PR to evaluate including it

closed 2023-07-10 by SergioGasquez