esp-rs/esp-template #103: cant build s3 template project

PocketPi, 2023-08-07:
I have generated a esp32 project with the esp32s3 model.

`
peter:/media/peter/data/code$ cargo generate esp-rs/esp-template
⚠️   Favorite `esp-rs/esp-template` not found in config, using it as a git repository: https://github.com/esp-rs/esp-template.git
🤷   Project Name: esp32s3-rust-test
🔧   Destination: /media/peter/data/code/esp32s3-rust-test ...
🔧   project-name: esp32s3-rust-test ...
🔧   Generating template ...
✔ 🤷   Configure advanced template options? · false
✔ 🤷   Which MCU to target? · esp32s3
🔧   Moving generated files into: `/media/peter/data/code/esp32s3-rust-test`...
Initializing a fresh Git repository
✨   Done! New project created /media/peter/data/code/esp32s3-rust-test
`

When i try to build it it fails: 

`peter:/media/peter/data/code/esp32s3-rust-test$ cargo build
    Updating crates.io index
   Compiling compiler_builtins v0.1.92
   Compiling core v0.0.0 (/home/peter/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core)
   Compiling proc-macro2 v1.0.66
   Compiling unicode-ident v1.0.11
   Compiling syn v1.0.109
   Compiling serde v1.0.183
   Compiling memchr v2.5.0
   Compiling rustversion v1.0.14
   Compiling serde_derive v1.0.183
   Compiling heck v0.4.1
   Compiling version_check v0.9.4
   Compiling autocfg v1.1.0
   Compiling anyhow v1.0.72
   Compiling quote v1.0.32
   Compiling aho-corasick v1.0.2
   Compiling regex-syntax v0.7.4
   Compiling lock_api v0.4.10
   Compiling xtensa-lx v0.8.0
   Compiling syn v2.0.28
   Compiling proc-macro-error-attr v1.0.4
   Compiling strsim v0.10.0
   Compiling fnv v1.0.7
   Compiling hashbrown v0.14.0
   Compiling strum v0.24.1
   Compiling ident_case v1.0.1
   Compiling equivalent v1.0.1
   Compiling proc-macro-error v1.0.4
   Compiling indexmap v2.0.0
   Compiling toml_datetime v0.6.3
   Compiling regex-automata v0.3.6
   Compiling winnow v0.5.4
   Compiling minijinja v0.15.0
   Compiling basic-toml v0.1.4
   Compiling regex v1.9.3
   Compiling darling_core v0.20.3
   Compiling toml_edit v0.19.14
   Compiling paste v1.0.14
   Compiling esp-synopsys-usb-otg v0.3.2
   Compiling log v0.4.18
   Compiling esp32s3 v0.19.0
   Compiling once_cell v1.18.0
   Compiling proc-macro-crate v1.3.1
error[E0463]: can't find crate for `unicode_ident` which `proc_macro2` depends on
   --> /home/peter/.cargo/registry/src/index.crates.io-6f17d22bba15001f/proc-macro-error-1.0.4/src/lib.rs:495:5
    |
495 |     pub extern crate proc_macro2;
    |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ can't find crate

error[E0463]: can't find crate for `proc_macro2`
   --> /home/peter/.cargo/registry/src/index.crates.io-6f17d22bba15001f/proc-macro-error-1.0.4/src/lib.rs:497:9
    |
497 |     use proc_macro2::Span;
    |         ^^^^^^^^^^^ can't find crate

error[E0463]: can't find crate for `proc_macro2`
   --> /home/peter/.cargo/registry/src/index.crates.io-6f17d22bba15001f/proc-macro-error-1.0.4/src/lib.rs:286:5
    |
286 | use proc_macro2::Span;
    |     ^^^^^^^^^^^ can't find crate

error[E0463]: can't find crate for `quote`
   --> /home/peter/.cargo/registry/src/index.crates.io-6f17d22bba15001f/proc-macro-error-1.0.4/src/lib.rs:287:5
    |
287 | use quote::{quote, ToTokens};
    |     ^^^^^ can't find crate

error[E0463]: can't find crate for `proc_macro2`
   --> /home/peter/.cargo/registry/src/index.crates.io-6f17d22bba15001f/proc-macro-error-1.0.4/src/dummy.rs:116:5
    |
116 | use proc_macro2::TokenStream;
    |     ^^^^^^^^^^^ can't find crate

error[E0463]: can't find crate for `proc_macro2`
 --> /home/peter/.cargo/registry/src/index.crates.io-6f17d22bba15001f/proc-macro-error-1.0.4/src/diagnostic.rs:2:5
  |
2 | use proc_macro2::Span;
  |     ^^^^^^^^^^^ can't find crate

error[E0463]: can't find crate for `proc_macro2`
 --> /home/peter/.cargo/registry/src/index.crates.io-6f17d22bba15001f/proc-macro-error-1.0.4/src/diagnostic.rs:3:5
  |
3 | use proc_macro2::TokenStream;
  |     ^^^^^^^^^^^ can't find crate

error[E0463]: can't find crate for `quote`
`

It works fine if I try to generate and build for the C3 model

SergioGasquez, 2023-08-08:
Hi! I assume you have properly [installed the `Xtensa` environment](https://esp-rs.github.io/book/installation/riscv-and-xtensa.html), do you mind sharing which `Xtensa` Rust version are you using?

SergioGasquez, 2023-08-10:
Are you sourcing the environment before building the project?

SergioGasquez, 2023-09-05:
Hello! I am closing this issue for inactivity, if the issue still persists, please feel free to reopen it.

closed 2023-09-05 by SergioGasquez