esp-rs/esp-template #110: Add support for generating embassy/async projects

MabezDev, 2023-09-07:
We should probably add an `embassy_main` macro for Xtensa to make this easier.

ctron, 2023-12-10:
I would love to have an example/template with async, embassy, and esp.

MabezDev, 2023-12-10:
We have plenty of async examples in esp-hal, and esp-wifi. As well as some projects in awesome-esp-rust. Would be nice to support generation here though, but it gets a bit tricky with the number of options we have.

ctron, 2023-12-11:
> We have plenty of async examples in esp-hal, and esp-wifi.

I was struggling a bit with the different options, compilation issues (around atomic-polyfil, portable-atomic, newer embassy versions, …). It took me a while, but in the end I got something going (with esp-hal from `main` branch). However, none of the examples seemed to work for me (due to various reasons).

I think it would be good to have a standalone example, which can be tested on a regular basis. And just forked/cloned to get started. And that feels pretty close to this repository.

tommasoclini, 2024-10-21:
Has progress been made on this topic? I'd really like an official template with embassy support out of the box.

SergioGasquez, 2024-10-22:
Hi! We are moving from `cargo-generate` to our own generation tool, see https://github.com/esp-rs/esp-template/issues/168. So, feel free to take a look at [`esp-generate`](https://github.com/esp-rs/esp-generate/), which supports embassy, but we are aware of [a bug when using embassy and wifi/ble](https://github.com/esp-rs/esp-generate/issues/17). 

The tool is new, so we appreciate any feedback if you end up ussing it!

tommasoclini, 2024-10-22:
Thank you, I'll take a look at it

closed 2024-11-14 by jessebraham