esp-rs/esp-template #115: Cannot flash ESP32 with web-flash tool

RuXBee, 2023-09-28:
I am getting started on the awesome world of embedded-rust, amazing!  :crab:  

I am not sure, but I didn't found any document related to `web-flash`.
https://github.com/esp-rs/esp-template/blob/ea343c0625ba4de99141c929324fad2964ec9b5f/scripts/flash.sh#L21

Also, in the __ESP Rust Book__ I found that the official tool used to flash ESP32 is `espflash`, that is based on `esptool.py`.
https://esp-rs.github.io/book/tooling/espflash.html#cargo-espflash
Fortunately, with this tool I could flash the ESP32 successfully.

Could anyone tell me why the template use this `web-flash` tool instead of `espflash`?
Thanks in advance! :smiley:

bjoernQ, 2023-09-29:
`web-flash` is for use in GitHub CodeSpaces or containers w/o access to the serial port.
The command just opens a socket and you could use a browser to flash the chip.

But if you are developing locally, you should use `espflash`. Actually, if you do `cargo run` the code generated by the template will use `espflash` as a runner

SergioGasquez, 2023-09-29:
Hi! I would recommend taking a look at https://esp-rs.github.io/book/writing-your-own-application/generate-project/esp-template.html, specially the [Running the Code](https://esp-rs.github.io/book/writing-your-own-application/generate-project/esp-template.html#running-the-code) section. Also, if you are into `no_std` make sure you check out the [`no_std` book](https://esp-rs.github.io/no_std-training/)

closed 2023-09-29 by SergioGasquez