esp-rs/esp-template #124: Cannot open linker script file linkall.x

oleid, 2023-11-05:
Dear all,

I'm trying to build a hello world project derived from this template on Windows 10 in a MSVC environment.
It was set up the following way:

1. Install rustup-msvc via scoop
2. Install microsoft command line compiler (like the scoop package suggested)
3. Install nightly rust via rustup
4. Install espup and cargo-generate via cargo
5. Run espup install according to documentation

After that I generated the project for esp32 via this git repo

`cargo build` compiles everything but linking fails with:

Running `C:\Users\Olaf\scoop\persist\rustup-msvc\.rustup\toolchains\esp\bin\rustc.exe --crate-name esp32_test --edition=2021 src\main.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --diagnostic-width=119 --crate-type bin --emit=dep-info,link -C embed-bitcode=no -C debuginfo=2 -C metadata=80f34878350df424 -C extra-filename=-80f34878350df424 --out-dir C:\Users\Olaf\src\esp32-test\target\xtensa-esp32-none-elf\debug\deps --target xtensa-esp32-none-elf -C incremental=C:\Users\Olaf\src\esp32-test\target\xtensa-esp32-none-elf\debug\incremental -L dependency=C:\Users\Olaf\src\esp32-test\target\xtensa-esp32-none-elf\debug\deps -L dependency=C:\Users\Olaf\src\esp32-test\target\debug\deps --extern noprelude:alloc=C:\Users\Olaf\src\esp32-test\target\xtensa-esp32-none-elf\debug\deps\liballoc-b83b4f40532317de.rlib --extern noprelude:compiler_builtins=C:\Users\Olaf\src\esp32-test\target\xtensa-esp32-none-elf\debug\deps\libcompiler_builtins-a6abaaf812b88ac3.rlib --extern noprelude:core=C:\Users\Olaf\src\esp32-test\target\xtensa-esp32-none-elf\debug\deps\libcore-20cc3db7174822c3.rlib --extern embedded_io=C:\Users\Olaf\src\esp32-test\target\xtensa-esp32-none-elf\debug\deps\libembedded_io-a11d11f01d50d33e.rlib --extern embedded_svc=C:\Users\Olaf\src\esp32-test\target\xtensa-esp32-none-elf\debug\deps\libembedded_svc-638243c65551179d.rlib --extern esp_alloc=C:\Users\Olaf\src\esp32-test\target\xtensa-esp32-none-elf\debug\deps\libesp_alloc-7a18a8bc1e9c1a16.rlib --extern esp_backtrace=C:\Users\Olaf\src\esp32-test\target\xtensa-esp32-none-elf\debug\deps\libesp_backtrace-ad8b19c56c9384b2.rlib --extern esp_println=C:\Users\Olaf\src\esp32-test\target\xtensa-esp32-none-elf\debug\deps\libesp_println-e2b09142032fb05f.rlib --extern esp_wifi=C:\Users\Olaf\src\esp32-test\target\xtensa-esp32-none-elf\debug\deps\libesp_wifi-59cc4053dbb9d2cc.rlib --extern hal=C:\Users\Olaf\src\esp32-test\target\xtensa-esp32-none-elf\debug\deps\libesp32_hal-bb25ebe35e864c4e.rlib --extern heapless=C:\Users\Olaf\src\esp32-test\target\xtensa-esp32-none-elf\debug\deps\libheapless-fbbf0ee7ccf78ebd.rlib --extern log=C:\Users\Olaf\src\esp32-test\target\xtensa-esp32-none-elf\debug\deps\liblog-7afd8e7f1ada38de.rlib --extern smoltcp=C:\Users\Olaf\src\esp32-test\target\xtensa-esp32-none-elf\debug\deps\libsmoltcp-5337ec9fc4f58151.rlib -Z unstable-options -C link-arg=-Tlinkall.x -C link-arg=-Trom_functions.x -C link-arg=-nostartfiles -L C:\Users\Olaf\src\esp32-test\target\xtensa-esp32-none-elf\debug\build\esp-wifi-sys-8b20ad09368be88d\out -L C:\Users\Olaf\src\esp32-test\target\xtensa-esp32-none-elf\debug\build\esp32-hal-c193d42e30ee7e61\out -L C:\Users\Olaf\src\esp32-test\target\xtensa-esp32-none-elf\debug\build\esp-hal-common-0291b71a3c0a8705\out -L C:\Users\Olaf\src\esp32-test\target\xtensa-esp32-none-elf\debug\build\esp32-8425ed446dd1dfa3\out -L C:\Users\Olaf\src\esp32-test\target\xtensa-esp32-none-elf\debug\build\xtensa-lx-833d551610d3c712\out -L C:\Users\Olaf\src\esp32-test\target\xtensa-esp32-none-elf\debug\build\xtensa-lx-rt-fad6f9f2fe77c11b\out`
[...]
`
```
 C:\Users\Olaf\scoop\persist\rustup-msvc\.rustup\toolchains\esp\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin/xtensa-esp-elf-ld.exe: cannot open linker script file linkall.x: No such file or directory
          collect2.exe: error: ld returned 1 exit status
````

Any idea what might be wrong here or any pointers how to debug this?

Thanks a lot,
Olaf

oleid, 2023-11-06:
> I would recommend trying to install Rust from the [official website](https://esp-rs.github.io/book/installation/rust.html), as using [package managers sometimes gives some troubles](https://www.rust-lang.org/tools/install). If you are targeting an `esp32`, which is Xtensa based, you don't need `nightly` toolchain. See https://esp-rs.github.io/book/installation/riscv-and-xtensa.html

Okay, I tried that. I 

* uninstalled `rustup-msvc` via scoop and deleted any remaining folder
* Installed rustup and rust-stable via official sources
* Installed espup and ran `espup install`
* Cleaned the existing build and ran `cargo build`.

 Sadly, it doesn't change anything:

````
  = note: C:/Users/Olaf/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: 
cannot open linker script file linkall.x: No such file or directory
          collect2.exe: error: ld returned 1 exit status
````

oleid, 2023-11-06:
Ah, I got it! Once I realized that the linker script got copied by esp32-hal's build script to the target out folder, I noticed that folder containing the rust project included an umlaut! Apparently gcc doesn't like umlauts on windows (doh!).

So should we document somewhere?

SergioGasquez, 2023-11-07:
Oh! Glad you found the issue, thanks for sharing the solution, just updated the title so any users that looks for the error can check the solution! Thanks again for the feedback!

closed 2023-11-07 by SergioGasquez