esp-rs/esp-template #125: Trouble running ESP32-C6 template

nbigaouette, 2023-11-09:
Got my hands on an ESP32-C6 and want to try rust on it. I am able to run the `esp-idf` examples ([hello_world](https://github.com/espressif/esp-idf/tree/master/examples/get-started/hello_world) and [blink](https://github.com/espressif/esp-idf/tree/master/examples/get-started/blink)) using `idf.py` to build and flash: I can see the hello world in my terminal or the del flash.

I'm having trouble running this template though.

I use `cargo generate esp-rs/esp-template` to generate a bare bone project:
```sh
❯ cargo generate esp-rs/esp-template
⚠️   Favorite `esp-rs/esp-template` not found in config, using it as a git repository: https://github.com/esp-rs/esp-template.git
🤷   Project Name: esp32-c6-hello-v5
🔧   Destination: esp32-c6-hello-v5 ...
🔧   project-name: esp32-c6-hello-v5 ...
🔧   Generating template ...
✔ 🤷   Which MCU to target? · esp32c6
✔ 🤷   Configure advanced template options? · true
✔ 🤷   Enable WiFi/Bluetooth/ESP-NOW via the esp-wifi crate? · false
✔ 🤷   Enable allocations via the esp-alloc crate? · false
✔ 🤷   Configure project to use Dev Containers (VS Code and GitHub Codespaces)? · false
✔ 🤷   Configure project to support Wokwi simulation with Wokwi VS Code extension? · false
✔ 🤷   Add CI files for GitHub Action? · false
✔ 🤷   Setup logging using the log crate? · false
🔧   Moving generated files into: `esp32-c6-hello-v5`...
🔧   Initializing a fresh Git repository
✨   Done! New project created esp32-c6-hello-v5
```

Building and flashing seems to succeed, but the monitoring hangs:
```sh
❯ cargo run -vv
[...]
     Running `CARGO=~/.rustup/toolchains/nightly-aarch64-apple-darwin/bin/cargo CARGO_MANIFEST_DIR=~/esp32-c6-hello-v4 CARGO_PKG_AUTHORS='' CARGO_PKG_DESCRIPTION='' CARGO_PKG_HOMEPAGE='' CARGO_PKG_LICENSE='MIT OR Apache-2.0' CARGO_PKG_LICENSE_FILE='' CARGO_PKG_NAME=esp32-c6-hello-v4 CARGO_PKG_README='' CARGO_PKG_REPOSITORY='' CARGO_PKG_RUST_VERSION='' CARGO_PKG_VERSION=0.1.0 CARGO_PKG_VERSION_MAJOR=0 CARGO_PKG_VERSION_MINOR=1 CARGO_PKG_VERSION_PATCH=0 CARGO_PKG_VERSION_PRE='' DYLD_FALLBACK_LIBRARY_PATH='~/esp32-c6-hello-v4/target/riscv32imac-unknown-none-elf/debug/build/esp-hal-common-4b0655f184eabea1/out:~/esp32-c6-hello-v4/target/riscv32imac-unknown-none-elf/debug/build/esp-riscv-rt-2022adf018f321c1/out:~/esp32-c6-hello-v4/target/riscv32imac-unknown-none-elf/debug/build/esp32c6-32bf530043214f0b/out:~/esp32-c6-hello-v4/target/riscv32imac-unknown-none-elf/debug/build/esp32c6-hal-03646f514ebc9229/out:~/esp32-c6-hello-v4/target/riscv32imac-unknown-none-elf/debug/deps:~/esp32-c6-hello-v4/target/riscv32imac-unknown-none-elf/debug:~/.rustup/toolchains/nightly-aarch64-apple-darwin/lib:~/lib:/usr/local/lib:/usr/lib' espflash flash --monitor target/riscv32imac-unknown-none-elf/debug/esp32-c6-hello-v4`
[2023-11-08T00:20:25Z INFO ] Detected 8 serial ports
[2023-11-08T00:20:25Z INFO ] Ports which match a known common dev board are highlighted
[2023-11-08T00:20:25Z INFO ] Please select a port
[2023-11-08T00:20:26Z INFO ] Serial port: '/dev/cu.usbmodem101'
[2023-11-08T00:20:26Z INFO ] Connecting...
[2023-11-08T00:20:27Z INFO ] Using flash stub
Chip type:         esp32c6 (revision v0.0)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi 6, BT 5
MAC address:       40:4c:ca:45:0a:cc
App/part. size:    177,392/4,128,768 bytes, 4.30%
[00:00:00] [========================================]      13/13      0x0
[00:00:00] [========================================]       1/1       0x8000
[00:00:01] [========================================]      70/70      0x10000
[2023-11-08T00:20:28Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x15 (USB_UART_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x408007fe
0x408007fe - software_reset_cpu
    at ??:??
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd48
0x4086c410 - __global_pointer$
    at ??:??
load:0x4086e610,len:0x2d68
0x4086e610 - __global_pointer$
    at ??:??
load:0x40875720,len:0x1800
0x40875720 - __global_pointer$
    at ??:??
SHA-256 comparison failed:
Calculated: aff89878a96cbff57c66c38aa6d1a422b7785040b5efea582babef6c357427c2
Expected: 0af544a033ab3492852b8232c904c578d5f07c9d4a423a64473f060db374ab32
Attempting to boot anyway...
entry 0x4086c410
0x4086c410 - __global_pointer$
    at ??:??
I (41) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (42) boot: compile time Jun  7 2023 08:02:08
I (43) boot: chip revision: v0.0
I (46) boot.esp32c6: SPI Speed      : 40MHz
I (51) boot.esp32c6: SPI Mode       : DIO
I (56) boot.esp32c6: SPI Flash Size : 4MB
I (61) boot: Enabling RNG early entropy source...
I (66) boot: Partition Table:
I (70) boot: ## Label            Usage          Type ST Offset   Length
I (77) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (84) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (92) boot:  2 factory          factory app      00 00 00010000 003f0000
I (99) boot: End of partition table
I (103) esp_image: segment 0: paddr=00010020 vaddr=42000020 size=23e80h (147072) map
I (142) esp_image: segment 1: paddr=00033ea8 vaddr=40808000 size=00020h (    32) load
I (143) esp_image: segment 2: paddr=00033ed0 vaddr=42023ed0 size=06598h ( 26008) map
I (153) esp_image: segment 3: paddr=0003a470 vaddr=40808020 size=01058h (  4184) load
I (157) boot: Loaded app from partition at offset 0x10000
I (162) boot: Disabling RNG early entropy source...
```

There's no output after the last line, no hello world :(

The expanded template result in:
```rust
#![no_std]
#![no_main]

use esp_backtrace as _;
use esp_println::println;
use hal::{clock::ClockControl, peripherals::Peripherals, prelude::*, Delay};

#[entry]
fn main() -> ! {
    let peripherals = Peripherals::take();
    let system = peripherals.SYSTEM.split();

    let clocks = ClockControl::max(system.clock_control).freeze();
    let mut delay = Delay::new(&clocks);

    println!("Hello world!");
    loop {
        println!("Loop...");
        delay.delay_ms(500u32);
    }
}
```
with the `.cargo/config.toml`:
```toml
[target.riscv32imac-unknown-none-elf]
runner = "espflash flash --monitor"

[build]
rustflags = [
  "-C", "link-arg=-Tlinkall.x",

  # Required to obtain backtraces (e.g. when using the "esp-backtrace" crate.)
  # NOTE: May negatively impact performance of produced code
  "-C", "force-frame-pointers",
]

target = "riscv32imac-unknown-none-elf"

[unstable]
build-std = ["core"]
```
and `Cargo.toml`:
```toml
[package]
name = "esp32-c6-hello-v4"
version = "0.1.0"
authors = [""]
edition = "2021"
license = "MIT OR Apache-2.0"

[dependencies]
hal = { package = "esp32c6-hal", version = "0.6.0" }
esp-backtrace = { version = "0.9.0", features = ["esp32c6", "panic-handler", "exception-handler", "print-uart"] }
esp-println = { version = "0.7.0", features = ["esp32c6"] }
```

Something I realized is that the built binary is large (larger than the available flash?):
```sh
❯ ls target/riscv32imac-unknown-none-elf/debug/esp32-c6-hello-v5
```
I tried `opt-level = "s"` and `opt-level = "z"` and `strip = true` which does reduce the size but the same result happens: there's no hello world :(

Any idea what I could be doing wrong? I'm not sure what to look for as playing with embedded is quite new for me.

Thanks!!

bjoernQ, 2023-11-09:
Probably your dev-board has two USB connectors, one named UART and one USB. the hello_world will print to UART so if you are using USB you won't see anything (after the boot message). You should see something after using the other connector.

Alternatively, you could run the `usb_serial_jtag.rs` example and you should see something printed

nbigaouette, 2023-11-09:
Thanks for the info!

Yes, the board has two USB-C connectors, one USB and one UART. I can't do anything (yet?) with the USB one, neither monitor nor flash: I simply cannot connect.

For example, if connected to the USB port:
```sh
❯ espflash monitor
[2023-11-09T13:49:05Z INFO ] Detected 8 serial ports
[2023-11-09T13:49:05Z INFO ] Ports which match a known common dev board are highlighted
[2023-11-09T13:49:05Z INFO ] Please select a port
✔ Remember this serial port for future use? · no
[2023-11-09T13:49:13Z INFO ] Serial port: '/dev/cu.usbmodem56CC0427961'
[2023-11-09T13:49:13Z INFO ] Connecting...
[2023-11-09T13:49:13Z INFO ] Using flash stub
Error:   × The bootloader returned an error
  ├─▶ Error while running MemData command
  ╰─▶ Other
```
```sh
❯ cargo run
    Finished dev [unoptimized + debuginfo] target(s) in 0.10s
     Running `espflash flash --monitor target/riscv32imac-unknown-none-elf/debug/esp32-c6-hello-v5`
[2023-11-09T13:50:59Z INFO ] Detected 8 serial ports
[2023-11-09T13:50:59Z INFO ] Ports which match a known common dev board are highlighted
[2023-11-09T13:50:59Z INFO ] Please select a port
✔ Remember this serial port for future use? · no
[2023-11-09T13:51:04Z INFO ] Serial port: '/dev/cu.usbmodem56CC0427961'
[2023-11-09T13:51:04Z INFO ] Connecting...
[2023-11-09T13:51:04Z INFO ] Using flash stub
Error:   × The bootloader returned an error
  ├─▶ Error while running MemData command
  ╰─▶ Other
```

Everything in the original post was obtained with the cable connected in the UART port, both flashing and monitor.

nbigaouette, 2023-11-09:
So I tried the [`esp32c6-hal`'s `usb_serial_jtag.rs` example](https://github.com/esp-rs/esp-hal/blob/c612fecfaef9f59473057d826d480785ec3fa513/esp32c6-hal/examples/usb_serial_jtag.rs).

I can see the hello world looping.

So the issue could be that the `println!()` output of the hello world example is not sent to the proper location? i.e. it should go to the "USB Serial/JTAG" but instead it goes somewhere else (where?). Am I getting this right?

Is there a way to use `println!()` or logging to be sent to the proper location on my board?

Thanks!

bjoernQ, 2023-11-10:
esp-println supports multiple "backends" chosen via features: https://github.com/esp-rs/esp-println/blob/c5bd57ab10ece3efa16e05d7a0d1322619ee64b8/Cargo.toml#L37-L42
(we need to improve documentation there, https://github.com/esp-rs/esp-println/issues/67)

When changing it you might also need to change it for esp-backtrace (https://github.com/esp-rs/esp-backtrace/blob/db2bb85bc9c24800287912efe305606a9f9ebb2a/Cargo.toml#L26-L30) which is slightly annoying t.b.h.

nbigaouette, 2023-11-11:
Awesome, using those dependencies:
```
[dependencies]
hal = { package = "esp32c6-hal", version = "0.6.0" }
esp-backtrace = { version = "0.9.0", features = [
    "esp32c6",
    "panic-handler",
    "exception-handler",
    # See https://github.com/esp-rs/esp-template/issues/125#issuecomment-1805242140
    # Don't use "print-uart" on the ESP32-C6, use this instead:
    "print-jtag-serial",
] }
esp-println = { version = "0.7.0", default-features = false, features = [
    "esp32c6",
    # See https://github.com/esp-rs/esp-template/issues/125#issuecomment-1805242140
    "jtag_serial",
    # Re-enable default features _except_ 'uart'
    "critical-section",
    "colors",
] }
```

the minimal hello world example worked as expected, thanks!!

closed 2023-11-11 by nbigaouette

will, 2024-05-12:
> With newer versions there is special option to print to jtag port. You should enable that feature for esp-println and disable default features which enables uart. So dependency will look like:
> 
> `esp-println = { version = "0.9.0", default-features = false, features = ["esp32c6", "jtag-serial", "log"] }`

This was the key I was missing. I wasn't sure the board was working at all until I tried making the builtin rgb light do something which led me to realize it was just debug printing that wasn't working.

Would it make sense to make it the default in the template? It would have solved me ~an hour of frustration, but I'm new to all of this so maybe other people figure it out quicker.

bjoernQ, 2024-05-14:
Ideally `esp-println` would get an option to auto-detect: https://github.com/esp-rs/esp-println/issues/86

I guess half of people will run into problems no matter which one we default to