esp-rs/esp-template #153: Porting a esp-hal example to a standalone esp-template generated repo?

brainstorm, 2024-03-24:
Hello!

I was trying to have one of the great embassy `esp-hal` examples as a standalone project that I wish to develop further but I ran into the following `rust-lld` issue (similar to https://github.com/esp-rs/esp-hal/issues/1336):

<description>**$ . ~/export-esp.sh**</description>
<details>$ env | grep ^PATH
PATH=/home/rvalls/.rustup/toolchains/esp/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin:/home/rvalls/.rustup/toolchains/esp/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin:/home/rvalls/.rustup/toolchains/esp/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin:/home/rvalls/.rustup/toolchains/esp/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin:/home/rvalls/.cargo/bin:/home/rvalls/.local/bin:/home/rvalls/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin:/home/rvalls/bin:/home/rvalls/.cargo/bin:/home/rvalls/.local/bin
</details>

<description>**$ espup update**</description>
<details><pre>
[2024-03-24T02:17:11Z WARN ] ⚠️   A new version of espup ('v0.11.0') is available
[2024-03-24T02:17:11Z INFO ] 💽  Updating Espressif Rust ecosystem
[2024-03-24T02:17:11Z INFO ] 💡  Querying GitHub API: 'https://api.github.com/repos/esp-rs/rust-build/releases/latest'
[2024-03-24T02:17:11Z INFO ] 💡  Querying GitHub API: 'https://api.github.com/repos/esp-rs/rust-build/releases'
[2024-03-24T02:17:12Z INFO ] 🔧  Checking Rust installation
[2024-03-24T02:17:12Z WARN ] ⚠️   Previous installation of LLVM exists in: '/home/rvalls/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-16.0.0-20230516'. Reusing this installation
[2024-03-24T02:17:12Z INFO ] 🔧  Installing RISC-V targets ('riscv32imc-unknown-none-elf' and 'riscv32imac-unknown-none-elf') for 'nightly' toolchain
[2024-03-24T02:17:12Z WARN ] ⚠️   Previous installation of GCC exists in: '/home/rvalls/.rustup/toolchains/esp/xtensa-esp32-elf/esp-12.2.0_20230208'. Reusing this installation
[2024-03-24T02:17:12Z WARN ] ⚠️   Previous installation of GCC exists in: '/home/rvalls/.rustup/toolchains/esp/xtensa-esp32s2-elf/esp-12.2.0_20230208'. Reusing this installation
[2024-03-24T02:17:12Z WARN ] ⚠️   Previous installation of GCC exists in: '/home/rvalls/.rustup/toolchains/esp/xtensa-esp32s3-elf/esp-12.2.0_20230208'. Reusing this installation
[2024-03-24T02:17:12Z WARN ] ⚠️   Previous installation of GCC exists in: '/home/rvalls/.rustup/toolchains/esp/riscv32-esp-elf/esp-12.2.0_20230208'. Reusing this installation
[2024-03-24T02:17:12Z WARN ] ⚠️   Previous installation of Xtensa Rust 1.76.0.1 exists in: '/home/rvalls/.rustup/toolchains/esp'. Reusing this installation
[2024-03-24T02:17:13Z INFO ] 🔧  Creating export file
[2024-03-24T02:17:13Z WARN ] 💡  Please, set up the environment variables by running: ' /home/rvalls/export-esp.sh'
[2024-03-24T02:17:13Z WARN ] ⚠️   This step must be done every time you open a new terminal
[2024-03-24T02:17:13Z INFO ] ✅  Update successfully completed!
</pre></details>

<description>**$ cargo generate esp-rs/esp-template**</description>
<details><pre>
⚠️   Favorite `esp-rs/esp-template` not found in config, using it as a git repository: https://github.com/esp-rs/esp-template.git
🤷   Project Name: esp-hal-embassy-i2s-example
🔧   Destination: /home/rvalls/dev/personal/pancake-legend-lights/src/esp-hal-embassy-i2s-example ...
🔧   project-name: esp-hal-embassy-i2s-example ...
🔧   Generating template ...
✔ 🤷   Which MCU to target? · esp32c6
✔ 🤷   Configure advanced template options? · true
✔ 🤷   Enable allocations via the esp-alloc crate? · false
✔ 🤷   Configure project to use Dev Containers (VS Code and GitHub Codespaces)? · true
✔ 🤷   Configure project to support Wokwi simulation with Wokwi VS Code extension? · true
✔ 🤷   Add CI files for GitHub Action? · true
✔ 🤷   Setup logging using the log crate? · false
✔ 🤷   Enable WiFi/Bluetooth/ESP-NOW via the esp-wifi crate? · true

For more information and examples of esp-wifi showcasing Wifi,BLE and ESP-NOW, see https://github.com/esp-rs/esp-wifi/blob/main/examples.md

🔧   Moving generated files into: `/home/rvalls/dev/personal/pancake-legend-lights/src/esp-hal-embassy-i2s-example`...
🔧   Initializing a fresh Git repository
✨   Done! New project created /home/rvalls/dev/personal/pancake-legend-lights/src/esp-hal-embassy-i2s-example
</pre></details>

Then I just went and copied the example file over to `main.rs` on the template (knowing that some error related to embassy not being supported in this template was about to come (see issue #110))... which I could just `cargo add` a posteriori (I guessed/thought):

**$ cp ~/dev/personal/esp-hal/examples/src/bin/embassy_i2s_read.rs src/main.rs**

But to my surprise I saw a `rust-lld` error instead:

<description>**$ cargo build**</description>
<details><pre>
    Updating crates.io index
   Compiling compiler_builtins v0.1.108
   Compiling core v0.0.0 (/home/rvalls/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core)
   Compiling proc-macro2 v1.0.79
(...)
   Compiling rustc-std-workspace-core v1.99.0 (/home/rvalls/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/rustc-std-workspace-core)
   Compiling stable_deref_trait v1.2.0
   Compiling critical-section v1.1.2
   Compiling embedded-hal v1.0.0
(...)
[... 348 lines cut ...]
got 4080 bytes, [7, 0, 0, 80, 7, 0, 0, 80, 4, 0]..[0, 80, 7, 0, 0, 80, 7, 0, 0, 80]
available 4080
got 4080 bytes, [4, 0, 0, 0, 4, 0, 0, 0, 4, 0]..[0, 80, 4, 0, 0, 0, 4, 0, 0, 0]
available 4080
got 4080 bytes, [4, 0, 0, 0, 4, 0, 0, 0, 7, 0]..[0, 0, 7, 0, 0, 80, 7, 0, 0, 80]
available 4080
got 4080 bytes, [7, 0, 0, 80, 7, 0, 0, 80, 8, 0]..[0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
available 4080
got 4080 bytes, [ff, ff, 0, 80, ff, ff, 0, 80, fc, ff]..[0, 0, 4, 0, 0, 0, 4, 0, 0, 0]
available 4080
got 4080 bytes, [7, 0, 0, 80, 7, 0, 0, 80, 4, 0]..[0, 0, 4, 0, 0, 0, 4, 0, 0, 0]
available 4080
got 4080 bytes, [4, 0, 0, 0, 4, 0, 0, 0, 4, 0]..[0, 0, 7, 0, 0, 80, 7, 0, 0, 80]
available 4080
got 4080 bytes, [7, 0, 0, 80, 7, 0, 0, 80, 7, 0]..[0, 0, 3, 0, 0, 80, 3, 0, 0, 80]
available 4080
got 4080 bytes, [3, 0, 0, 80, 3, 0, 0, 80, 0, 0]..[0, 80, 4, 0, 0, 0, 4, 0, 0, 0]
available 4080
got 4080 bytes, [ff, ff, 0, 80, ff, ff, 0, 80, 0, 0]..[0, 0, fc, ff, 0, 0, fc, ff, 0, 0]
(...) CTRL-C
</pre></details>

For context I'm just trying to pipe I2S mems microphone data, see https://github.com/esp-rs/esp-println/issues/29#issuecomment-2016655068.

What I am doing wrong here w.r.t the `rust-lld -nostartfiles` error?

SergioGasquez, 2024-03-25:
As mentioned in https://github.com/esp-rs/esp-hal/issues/1336#issuecomment-2014543276, looks like you are missing some `.cargo/config.toml` configurations. 

Feel free to reopen the issue if you consider it is not completed.

brainstorm, 2024-03-25:
@SergioGasquez Not the same issue, here's my `.cargo/config.toml`:

```toml
[target.riscv32imac-unknown-none-elf]
runner = "espflash flash --monitor"

[build]
rustflags = [
  "-C", "link-arg=-Tlinkall.x",

  # Required to obtain backtraces (e.g. when using the "esp-backtrace" crate.)
  # NOTE: May negatively impact performance of produced code
  "-C", "force-frame-pointers",
]

target = "riscv32imac-unknown-none-elf"

[unstable]
build-std = ["core"]
```

could you please point out what I'm missing?

bjoernQ, 2024-03-25:
There is something strange going on.

`-nostartfiles` should only get added to `.cargo/config.toml` for Xtensa: https://github.com/esp-rs/esp-template/blob/21b13bc6089ef67d65e869b7fd03545de3289eff/.cargo/config.toml#L16-L18

And in the posted file above it's not present - so the big question would be what adds that option?

brainstorm, 2024-03-25:
Since I cannot tell if the last question is aimed at me, I re-did the steps above right now from scratch just in case :P ... and got another `rust-lld` missing file, the `rom_functions.x` one too this time:

```
$ cargo generate esp-rs/esp-template
⚠️   Favorite `esp-rs/esp-template` not found in config, using it as a git repository: https://github.com/esp-rs/esp-template.git
🤷   Project Name: rust
🔧   Destination: /home/rvalls/dev/personal/pancake-legend-lights/src/rust ...
🔧   project-name: rust ...
🔧   Generating template ...
✔ 🤷   Which MCU to target? · esp32c6
✔ 🤷   Configure advanced template options? · false
🔧   Moving generated files into: `/home/rvalls/dev/personal/pancake-legend-lights/src/rust`...
🔧   Initializing a fresh Git repository
✨   Done! New project created /home/rvalls/dev/personal/pancake-legend-lights/src/rust
```

Here's the generated `.cargo/config.toml`

```
$ cat .cargo/config.toml 
[target.riscv32imac-unknown-none-elf]
runner = "espflash flash --monitor"

[build]
rustflags = [
  "-C", "link-arg=-Tlinkall.x",

  # Required to obtain backtraces (e.g. when using the "esp-backtrace" crate.)
  # NOTE: May negatively impact performance of produced code
  "-C", "force-frame-pointers",
]

target = "riscv32imac-unknown-none-elf"

[unstable]
build-std = ["core"]
```

Here's the end of `cargo build` (also fails the same way with `--release`):

```
(...)
  = note: rust-lld: error: unknown argument '-nostartfiles'
          rust-lld: error: cannot find linker script rom_functions.x

```

Are any of you two able to reproduce this issue or it's just me?

bjoernQ, 2024-03-25:
I'm not able to reproduce it given the steps above. But since it seems cargo is picking up something which is not in your generated `.cargo/config.toml` maybe it's picking up something from another file: https://doc.rust-lang.org/cargo/reference/config.html - e.g. having a `config.toml` in `$CARGO_HOME` or in any of the other options provided

brainstorm, 2024-03-25:
> cargo -Z unstable-options config get

Thanks @bjoernQ, here it is:

```
$ cargo -Z unstable-options config get
build.rustflags = ["-C", "link-arg=-Tlinkall.x", "-C", "link-arg=-Trom_functions.x", "-C", "link-arg=-nostartfiles", "-C", "link-arg=-Tlinkall.x", "-C", "force-frame-pointers"]
build.target = "riscv32imac-unknown-none-elf"
env.ESP_LOGLEVEL = "INFO"
target.riscv32imac-unknown-none-elf.runner = "espflash flash --monitor"
target.xtensa-esp32s3-none-elf.runner = "espflash flash --monitor"
unstable.build-std = ["core", "core"]
# The following environment variables may affect the loaded values.
# CARGO_HOME=/home/rvalls/.cargo
```

On a quick look, I don't see any config overrides/definitions in `/home/rvalls/.cargo`:

```
$ cat ~/.cargo/
bin/                   .crates.toml           git/                   .package-cache         registry/              
.crates2.json          env                    .global-cache          .package-cache-mutate  
rvalls@framework:~/dev/personal/pancake-legend-lights/src/rust$ cat ~/.cargo/.crates.toml 
[v1]
"binocle 0.3.1 (registry+https://github.com/rust-lang/crates.io-index)" = ["binocle"]
"cargo-generate 0.19.0 (registry+https://github.com/rust-lang/crates.io-index)" = ["cargo-generate"]
"du-dust 0.8.6 (registry+https://github.com/rust-lang/crates.io-index)" = ["dust"]
"espflash 2.1.0 (registry+https://github.com/rust-lang/crates.io-index)" = ["espflash"]
"espup 0.6.1 (registry+https://github.com/rust-lang/crates.io-index)" = ["espup"]
"just 1.23.0 (registry+https://github.com/rust-lang/crates.io-index)" = ["just"]
"qrrs 0.1.9 (registry+https://github.com/rust-lang/crates.io-index)" = ["qrrs"]
"ravedude 0.1.8 (registry+https://github.com/rust-lang/crates.io-index)" = ["ravedude"]
rvalls@framework:~/dev/personal/pancake-legend-lights/src/rust$ cat ~/.cargo/env 
#!/bin/sh
# rustup shell setup
# affix colons on either side of $PATH to simplify matching
case ":${PATH}:" in
    *:"$HOME/.cargo/bin":*)
        ;;
    *)
        # Prepending path in case a system-installed rustc needs to be overridden
        export PATH="$HOME/.cargo/bin:$PATH"
        ;;
esac

rvalls@framework:~/dev/personal/pancake-legend-lights/src/rust$ env | grep -i cargo
PATH=/home/rvalls/.rustup/toolchains/esp/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin:/home/rvalls/.rustup/toolchains/esp/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin:/home/rvalls/.rustup/toolchains/esp/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin:/home/rvalls/.rustup/toolchains/esp/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin:/home/rvalls/.cargo/bin:/home/rvalls/.local/bin:/home/rvalls/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin:/home/rvalls/bin:/home/rvalls/.cargo/bin:/home/rvalls/.local/bin
```

Thanks for the assistance either way!

bjoernQ, 2024-03-25:
Just for reference this is what the effective config looks for me
```
❯ cargo -Z unstable-options config get
build.rustflags = ["-C", "link-arg=-Tlinkall.x", "-C", "link-arg=-Trom_functions.x", "-C", "force-frame-pointers"]
build.target = "riscv32imac-unknown-none-elf"
target.riscv32imac-unknown-none-elf.runner = "espflash flash --monitor"
unstable.build-std = ["core"]
# The following environment variables may affect the loaded values.
# CARGO_HOME=C:\Users\Bjoern\.cargo
```

e.g. `target.xtensa-esp32s3-none-elf.runner = "espflash flash --monitor"` looks suspicious and also the problematic `"-C", "link-arg=-nostartfiles",` is there

`cargo -Z unstable-options config get --show-origin` might help .... it should show where each setting is from

brainstorm, 2024-03-25:
Aha! Thanks so much, there was this remaining `.cargo` leftover from a previous template generation **on the top level of the project** (I was generating sub-rust projects and posting the sub-.cargo dirs), I didn't know that top level `.cargo` (not $HOME/.cargo, but repo top level) took precedence, interesting behavior!:

```
$ cat .cargo/config.toml 
[target.xtensa-esp32s3-none-elf]
runner = "espflash flash --monitor"

[env]
ESP_LOGLEVEL="INFO"
[build]
rustflags = [
  "-C", "link-arg=-Tlinkall.x",
"-C", "link-arg=-Trom_functions.x",

  "-C", "link-arg=-nostartfiles",
]

target = "xtensa-esp32s3-none-elf"

[unstable]
build-std = ["core"]
```

Thanks so much for the help, apologies for the confusion!

closed 2024-03-25 by brainstorm