esp-rs/esp-template #155: Why does esp-template use v0.2.7 of embedded-hal instead of expected v1.0.0?

henryshunt, 2024-03-27:
If I create a new project for the ESP32-S3 using `cargo generate esp-rs/esp-template` and run `cargo tree`, I see dependencies on v0.2.7 of `embedded-hal` by `esp-hal` v0.16.1 and `esp-synopsys-usb-otg` v0.4.0. I had expected there to be only dependencies on v1.0.0 of `embedded-hal` given that from reading some of the release notes it appears that v1.0.0 is properly supported (and thus I assume it is supposed to be the default). Additionally, `esp-hal`'s `Cargo.toml` file specifies that it should use v1.0.0 (except for the `embedded-hal-02` case to support the old version of course).

I'm trying to use v1.0.0 of `embedded-hal` in an S3 project I'm working on, but this is causing a compilation error when I work with GPIO in a certain way. I believe this is caused by this version mismatch.

I can post the error when I get home if necessary, but I'm primarily just trying to understand where the old version of `embedded-hal` is coming from given that from what I can tell, it should be using v1.0.0.

SergioGasquez, 2024-04-02:
Some drivers are still using `embedded-hal 0.2.7`, see https://github.com/esp-rs/esp-hal/issues/1317. Also note that some drivers are already merged (https://github.com/esp-rs/esp-hal/issues/1317#issuecomment-2025401755) are not yet released.

henryshunt, 2024-04-09:
Ah, well that explains things then. I was under the mistaken impression that the v1.0.0 support was already completed. Thanks!

closed 2024-04-09 by henryshunt