esp-rs/esp-template #172: error : for $ idf.py menuconfig command

TGG-3, 2024-09-12:
I extracted xtensa-esp32-elf-linux32-1.22.0-80-g6c4433a-5.2.0 into downloads folder and trying to set target device that shown an error again tried to use menuconfig command the below error occured
Executing action: menuconfig
Running cmake in directory /home/g3/esp/hello_world/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home/g3/.espressif/python_env/idf5.2_py3.10_env/bin/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/g3/esp/hello_world"...
-- IDF_TARGET not set, using default target: esp32
-- Found Git: /usr/bin/git (found version "2.34.1")
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is unknown
-- Found assembler: /home/g3/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /home/g3/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/g3/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - broken
CMake Error at /usr/share/cmake-3.22/Modules/CMakeTestCCompiler.cmake:69 (message):
  The C compiler

    "/home/g3/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/g3/esp/hello_world/build/CMakeFiles/CMakeTmp

    Run Build Command(s):/usr/bin/ninja cmTC_e81de && [1/2] Building C object CMakeFiles/cmTC_e81de.dir/testCCompiler.c.obj
    FAILED: CMakeFiles/cmTC_e81de.dir/testCCompiler.c.obj
    /home/g3/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc   -mlongcalls -Wno-frame-address -o CMakeFiles/cmTC_e81de.dir/testCCompiler.c.obj -c /home/g3/esp/hello_world/build/CMakeFiles/CMakeTmp/testCCompiler.c
    /bin/sh: 1: /home/g3/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: not found
    ninja: build stopped: subcommand failed.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  /home/g3/esp-idf/tools/cmake/project.cmake:506 (__project)
  CMakeLists.txt:6 (project)

bjoernQ, 2024-09-12:
I'm not sure I understand what you are trying to do but this is the template repository for bare-metal Rust

If you are looking for the Rust on top of esp-idf template you might want to look into: https://github.com/esp-rs/esp-idf-template
If you are looking for help with ESP-IDF: https://github.com/espressif/esp-idf

closed 2024-09-12 by bjoernQ

TGG-3, 2024-09-12:
thank you @bjoernQ I was trying to build a ESP project in WSL and received xtensa-esp32-elf-gcc error but as suggested here https://github.com/espressif/esp-idf, after running  install.sh
source export.sh this error was cleared thank you