esp-rs/esp-template #175: probe-rs debugging for esp-no-std debugging (esp32c3)

Krensi, 2024-10-15:
Hi!

I just cannot figure out, how to debug my esp32c3 using probe-rs. 
I used the (template)[https://github.com/esp-rs/esp-template/blob/main/src/main.rs] and configured it to include the wifi snippet. If I run the main function using the "probe-rs" runner, everything works fine.
However, if I use `launch.json` to start debugging I always get a memory error. I suspect, that I overwrite a wrong sector or something. 

Could someone help me with a `launch.json` file?

If I add 
``` toml
[target.'cfg(any(target_arch = "riscv32", target_arch = "xtensa"))']
runner = "probe-rs run --chip esp32c3"
```
to `config.toml` I can run the main function without any issues. So probe-rs works basically.

Unfortunately the [example](https://docs.esp-rs.org/book/tooling/debugging/probe-rs.html#example-launchjson) does not work for me. 

I have the following launch.json

``` json
{
    "version": "0.2.0",
    "configurations": [
        {
            "type": "probe-rs-debug",
            "request": "launch",
            "name": "Launch",
            "cwd": "${workspaceFolder}",
            "chip": "esp32c3",
            "flashingConfig": {
                "flashingEnabled": true,
                "haltAfterReset": false,
                "formatOptions": {
                    "binaryFormat": "idf"
                }
            },
            "coreConfigs": [
                {
                    "coreIndex": 0,
                    "programBinary": "target/riscv32imc-unknown-none-elf/debug/${workspaceFolderBasename}",
                    "svdFile": "${workspaceFolder}/esp32c3.svd"
                }
            ]
        },
        {
            "type": "probe-rs-debug",
            "request": "attach",
            "name": "Attach",
            "cwd": "${workspaceFolder}",
            "chip": "esp32c3",
            "coreConfigs": [
                {
                    "coreIndex": 0,
                    "programBinary": "target/riscv32imc-unknown-none-elf/debug/${workspaceFolderBasename}",
                    "svdFile": "${workspaceFolder}/esp32c3.svd"
                }
            ]
        }
    ]
}
```

This is the error message I am getting after using `debug` with `launch.json`

```
====================== PANIC ======================
panicked at src/main.rs:39:6:
called `Result::unwrap()` on an `Err` value: WifiError(InternalError(EspErrNoMem))

Backtrace:

0x42000434
0x4200637e
0x42000132
```

Thank you for your help!

Greetings, Christian

SergioGasquez, 2024-10-18:
Hi! Just tried reproducing your issue with no luck, here is the template options that I used:
```
⚠️   Favorite `esp-rs/esp-template` not found in config, using it as a git repository: https://github.com/esp-rs/esp-template.git
🤷   Project Name: issue175
🔧   Destination: /Users/sergio/Documents/Espressif/tests/issue175 ...
🔧   project-name: issue175 ...
🔧   Generating template ...
✔ 🤷   Which MCU to target? · esp32c3
✔ 🤷   Configure advanced template options? · true
✔ 🤷   Enable allocations via the esp-alloc crate? · true
✔ 🤷   Enable WiFi/Bluetooth/ESP-NOW via the esp-wifi crate? · true
✔ 🤷   Configure project to use Dev Containers (VS Code and GitHub Codespaces)? · false
✔ 🤷   Configure project to support Wokwi simulation with Wokwi VS Code extension? · false
✔ 🤷   Add CI files for GitHub Action? · false
```

Then I updated the runner in `.cargo/config.toml` to `runner = "probe-rs run"` and copied your `launch.json`. I set a breakpoint on main() and I was able to launch the debugger and stop there without any issue.

Im using `probe-rs 0.24.0 (git commit: 9bde591c)`

By the way, for questions like this one, Id recommend using our matrix channel: https://matrix.to/#/#esp-rs:matrix.org

Krensi, 2024-10-18:
@SergioGasquez thank you!

Yes I managed to do this also sometimes. But can you also initialize the WiFi? I always get the memory error. So I figured, that there might be some issue with overwriting the WiFi related esp sector or something.

When I hit continue and want to pass the WiFi init I always get a panic.
I will try downloading the template again and I also make sure to have the exact same version of `probe-rs` as you.

Krensi, 2024-10-22:
This issue should be solved by https://github.com/esp-rs/esp-template/issues/176

closed 2024-10-22 by Krensi