esp-rs/esp-template #178: Archive this repository in favour of using `esp-generate`

jessebraham, 2024-11-08:
With the first version of [esp-generate](https://github.com/esp-rs/esp-generate/) published, we should focus our efforts there rather than continuing development of this template.

Given the newness of the tool we should probably wait a bit longer before archiving this repository, but I don't think it's worth investing any more development time here, and as such I'm opening this issue as a warning to potential new users.

Once we're convinced that `esp-generate` is mature enough to promote as our officially recommended solution, we can archive this repository.

jessebraham, 2024-11-14:
Created issues is the relevant repositories for a couple these:

- https://github.com/esp-rs/book/issues/171
- https://github.com/esp-rs/no_std-training/issues/61

Going to open a PR updating the docs in the HAL shortly here, and @SergioGasquez already updated our awesome list.

SergioGasquez, 2024-11-14:
I'll work on updating the books!

PR #179 (2024-11-14): Add deprecation notice to `README.md`
Are we happy with the text I've added? Will make any changes if needed, of course.

Closes #178, will archive the repo once this is merged.
closed 2024-11-14 by jessebraham